outdoorbits/little-backup-box #214: Thumbnails/Email

thelightdecides, 2023-07-03:
Hi,
I have two problems.
1.
I have disabled thumbnail creation under settings. Nevertheless LBB starts the generation of thumbnail after the copy process is finished.

2.
I didn't get a mail after  backup has finished. Test email was send without problems.

outdoorbits, 2023-07-03:
I'll check that, but will need a few days. Could you provide a log file, loglevel to maximum?

outdoorbits, 2023-07-03:
1. Please have a second look: Did you disable thumbnail creation for 
automatic standard backup (Backup - Default backup mode - Settings for 
the primary standard backup). This affects the default backup only. Did 
you set it in the settings in "Image viewer View" (more below) as 
standard?

2. Did you enable email notifications in "Email Notifications" section?

thelightdecides, 2023-07-03:
Thank you for your quick reply.
I forgot to disable it under "Backup", I had only disabled the settings under "Image Viewer View".
Now everything works fine. I also received the email after the backup was complete. Thanks for the help!

closed 2023-07-07 by outdoorbits