outdoorbits/little-backup-box #222: Big Foto Files Break down

Stephan1958, 2023-07-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

outdoorbits, 2023-07-15:
Did you backup from camera or cardreader? At first I don't have an idea, please give more information.

AoXis, 2023-08-10:
Have you tried to disable the preview?

closed 2023-09-01 by outdoorbits