outdoorbits/little-backup-box #223: sshd doesn't work after clean install

dietrest, 2023-08-02:
I've installed on fresh raspi os 64 lite image and signed th wlan up to my Cellphon hotspot. I can access the web gui of the lbb but ssh connection doesn't work anymore.
When I connect hdmi display and keyboard I can login with pi/raspberry.

but systemctl status sshd complains about Permissions 0777 for /etc/ssh/ssh_host_ecdsa_key and /etc/ssh/ssh_host_ed25519_key  are too open. 

Also sudo does not work :  /etc/sudo.conf  is world writeable , /usr/bin/sudo must be owned by uid 0 and have the setuid bit set.

So no updates, raspi-conf, ssh  possible. :-((

outdoorbits, 2023-08-03:
This is maybe an issue to comitup. I'm in holiday, so I can't test or help. Anybody else?

dietrest, 2023-08-03:
I've reinstalled from sratch and said no to install wifi hotspot with comitup and now everything seems to work as it should do.
The files in /usr/bin have 0755. When I have said yes to install comitup, every file in /usr/bin get 0777 and in /etc the same.
After that no ssh no samba no wifi no sudo no update ........

Dan-northwind, 2023-08-05:
I am seeing the same, if I install on an SSD it causes the issues above, if I install on a SD card it seems to work ok, either with or without comitup.

dietrest, 2023-08-08:
Yes, my problems too occured when trying to install on m.2 ssd in the box connected to USB

Dan-northwind, 2023-08-08:
What I have discovered is that it only works from the sd card, when you try and boot using an ssd, the script then tries to mount the ssd on boot as a storage device and this messes the permissions up on the drive.

closed 2023-08-09 by dietrest