outdoorbits/little-backup-box #234: "Power down after idle-time" does not work

chatulim, 2023-09-04:
**Describe the bug**
If no file transfer is initiated, LBB does not power down automatically at the specified -- no matter what is selected in the "Behavior" setting.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to http://lbbox.local:8000/config.php
2. In "Behavior" setting, select Power down after idle-time of "2 minutes" (or "5 minutes" or whatever)
3. Save settings (confirm "config.cfg was written successfully.")
4. Reboot
5. Wait allotted time (without attaching a source USB device)
6. LBB does not shut down

**Expected behavior**
In older versions of LBB, the RPi would just power off after the set idle time. That does not happen with the current version. It needs to be shut down manually from the command line or the web interface. .

**Screenshots**
[image]

**Desktop (please complete the following information):**
- Pi Zero 2 W
- Raspberry Pi OS Lite (Bullseye)

outdoorbits, 2023-09-04:
Did you close the web UI? It reloads log and keeps it active.

chatulim, 2023-09-04:
Yes. I closed the web UI. Sorry I did not put that into the initial report. I also tried powering the RPi down from the web UI, then closing the webpage, then cycling the power on the RPi to boot it fresh with no web UI or SSH interaction at all. 

Same result -- the RPi just stays on until I go in (via web UI or SSH from the Terminal of my Mac) and shut it down manually.

outdoorbits, 2023-09-04:
Are you on main or development branch?

outdoorbits, 2023-09-04:
Is there a standard backup configured and LBB is waiting for a device?

chatulim, 2023-09-04:
I am on the main branch. There is a standard backup configured (USB -> USB).

Below is the contents of the downloaded settings (with password obscured, no other changes made):

> conf_LANGUAGE=''
> conf_BACKUP_DEFAULT_SOURCE='usb'
> conf_BACKUP_DEFAULT_TARGET='usb'
> conf_BACKUP_DEFAULT_GENERATE_THUMBNAILS=false
> conf_BACKUP_DEFAULT_UPDATE_EXIF=false
> conf_BACKUP_DEFAULT_SOURCE2='none'
> conf_BACKUP_DEFAULT_TARGET2='none'
> conf_BACKUP_CAMERA_FOLDER_MASK='*:DCIM'
> conf_BACKUP_TARGET_BASEDIR_CLOUD='little-backup-box'
> conf_POWER_OFF=true
> conf_DISP=false
> conf_DISP_CONNECTION='I2C'
> conf_DISP_DRIVER='SSD1306'
> conf_DISP_I2C_ADDRESS='3c'
> conf_DISP_SPI_PORT='0'
> conf_DISP_RESOLUTION_X=128
> conf_DISP_RESOLUTION_Y=64
> conf_DISP_ROTATE='0'
> conf_DISP_CONTRAST=255
> conf_DISP_COLOR_MODEL='1'
> conf_DISP_COLOR_TEXT='white'
> conf_DISP_COLOR_HIGH='yellow'
> conf_DISP_COLOR_ALERT='red'
> conf_DISP_FONT_SIZE=12
> conf_DISP_FRAME_TIME=2
> conf_DISP_BLACK_ON_POWER_OFF=true
> conf_DISP_IP_REPEAT=true
> conf_MENU_ENABLED=true
> conf_MENU_BUTTON_COMBINATION='1'
> conf_MENU_BUTTON_BOUNCETIME=100
> conf_FAN_PWM_TEMP_C=0
> conf_FAN_PWM_GPIO=18
> conf_THEME=dark
> conf_BACKGROUND_IMAGE=
> conf_POPUP_MESSAGES=true
> conf_LOGLEVEL=1
> conf_LOG_SYNC=true
> conf_POWER_OFF_IDLE_TIME=2
> conf_BACKUP_GENERATE_THUMBNAILS=false
> conf_BACKUP_UPDATE_EXIF=false
> conf_VIEW_CONVERT_HEIC=false
> conf_VIEW_WRITE_RATING_EXIF=false
> conf_MAIL_NOTIFICATIONS=false
> conf_MAIL_HTML=false
> conf_SMTP_SERVER=''
> conf_SMTP_PORT='465'
> conf_MAIL_SECURITY='STARTTLS'
> conf_MAIL_USER=''
> conf_MAIL_PASSWORD=''
> conf_MAIL_TO=''
> conf_RSYNC_SERVER=''
> conf_RSYNC_PORT='873'
> conf_RSYNC_USER=''
> conf_RSYNC_PASSWORD=''
> conf_RSYNC_SERVER_MODULE='little-backup-box'
> conf_WIFI_COUNTRY='IL'
> conf_VPN_TYPE_RSYNC='none'
> conf_VPN_TYPE_CLOUD='none'
> conf_VPN_TIMEOUT='20'
> conf_PASSWORD="****************"
> conf_MAIL_FROM=''

outdoorbits, 2023-09-04:
I can't reproduce that. But maybe you just switch to development version. It's close to replace main and my tests on RPI4 where very successful.

chatulim, 2023-09-06:
I tried testing again -- this time with the development branch of LBB on a fresh install of Pi OS (Bullseye)-- and can confirm that "Power down after idle-time" does not work as expected on a Pi Zero W, a Pi Zero 2 W, or a Pi 3B+. On all 3 devices, if no transfer was initiated, the RPi would stay on until manually shut down even when the "idle-time" setting was set to power down after a specific number of minutes.

Are you testing on any model other than the Raspberry Pi 4?

outdoorbits, 2023-09-06:
I'm just testing on my RP 4 and actually I even don't have an alternative device. But in principle there shouldn't be an difference here. Could you please check the following:

From terminal on your pi type
crontab -e

Here should be a line containing cron_idletime. Please copy this command and execute it. Do you get an error message?

chatulim, 2023-09-08:
Running `crontab -e` generates this:

@reboot sudo python3 /var/www/little-backup-box/backup-autorun.py
@reboot sudo python3 /var/www/little-backup-box/start-rclone-gui.py
*/1 * * * * sudo python3 /var/www/little-backup-box/cron_ip.py
*/1 * * * * sudo python3 /var/www/little-backup-box/cron_idletime.py

Running `sudo python3 /var/www/little-backup-box/cron_idletime.py` generates... nothing. No error, no indication that anything happened, just another prompt — ***@***.***:~ $` — and a gray cursor.

outdoorbits, 2023-09-08:
That means, cron job is configured and the process does not raise an error. Everything we expected. Is this an absolute native installation? Any changes?

chatulim, 2023-09-12:
I’m not sure what you mean by “absolute native installation”. I do not think I am making any drastic changes to either the OS or LBB. Here is what I do each time I run a test:

- Fresh install of Pi OS (Bullseye Lite 32 bit) onto Micro SD card using Raspberry Pi Imager on an M1 Mac
- Login to Pi via SSH from Mac Terminal
- Fresh install of LBB —I tried installing both the main and development branch, both with and without Comitup
- Open settings in web view and change settings for “Backup” and "Behavior” to power down (after backup and if idle for 5 minutes). 

I don't change too many other settings, although there are a few other settings that have to be changed with every fresh install -- like time-zone and WiFi country. In theory those were alread set correctly as part of the OS install. I don't know why those settings revert to default...

outdoorbits, 2023-09-12:
O.k., sometimes we have been hunting for problems and after wasting a lot of time some major modifications came up - unfortunately this is not the case here... ;-)
I can not explain, why it does not shutdown. Idle time means no rsync, no gphoto2 and no web-access. I would like to send you a short script to find the problem. Maybe tonight...
The lost configuration is a bug, thank you! I will try to catch it.

outdoorbits, 2023-09-12:
I just tried to reproduce the config issue after update but everything was in place. Can you reproduce this behavior always after updates? Do you update by web UI?

outdoorbits, 2023-09-12:
Could you please update. After reboot on terminal, please try this command:
`sudo python3 /var/www/little-backup-box/cron_idletime.py`
What does it print?

chatulim, 2023-09-13:
> I just tried to reproduce the config issue after update but everything was in place. Can you reproduce this behavior always after updates? Do you update by web UI?

Until now I wouldn't have known if the configuration issue comes up after updates because until now I have never *updated* LBB. Never. Not at all. **Every time** I would try to test it, I would wipe the SD card and do a clean-install of the OS *and* a clean-install of LBB. And that was when I saw the problem:

In the process of doing a clean install of Pi OS, I would use the Raspberry Pi Imager app to customize some key settings -- including the **WiFi Country** and **Time zone**. I set them for my locale before I even install LBB. Then, when I do install LBB (either the main or development branch), I would expect these settings to be the way I had them, but on the LBB settings page they are set to "Great Britain" and "London". That is the problem I was referring to.

As far as *updating*... I just ran an update -- using the web UI -- and it looks like everything stayed the way it should be. So at least that is not a problem.

chatulim, 2023-09-13:
> Could you please update. After reboot on terminal, please try this command:
> `sudo python3 /var/www/little-backup-box/cron_idletime.py`
> What does it print?

I ran the latest update and ran the command. It returned the following:

LbbLogfileAgeSec=42.03113651275635s
ApacheLogfileAgeSec=1.9806199073791504s

After a while I realized that I still had the Web UI open, so I **closed the Web UI**, rebooted the Pi again, and ran the command again. This time it returned this:

`LbbLogfileAgeSec=26.245579957962036s
`ApacheLogfileAgeSec=1519.9275805950165s

I have no idea what this means. Is it helpful to you?

outdoorbits, 2023-09-13:
It is. LBB creates a log file for its activities. This file aged 26 seconds while your last web UI access was ist than 1500 s. That means, the are activities logged where nothing should happen. Could you please download the log file and tell me about the last message?

chatulim, 2023-09-13:
Here is the contents of the log file after letting it run for a few minutes (not the full 5 minutes):

```
Little Backup Box
2023-09-13 12:39:37
IMAGE=/var/www/little-backup-box/little-backup-box.bmp

2023-09-13 12:39:40
Int. storage o.k.
size: 30G
used: 3G
free: 27G
FS: ext4
 8.3% >___________________

2023-09-13 12:39:40
mount device: usb, source; WaitForDevice=True; DeviceIdentifierPresetThis=; DeviceIdentifierPresetOther=

2023-09-13 12:39:40
Ready
Insert source

2023-09-13 12:40:08
IP online
192.168.1.113

2023-09-13 15:58:53
IP online
192.168.1.113

2023-09-13 15:59:03
IP online
192.168.1.113

2023-09-13 16:00:03
IP online
192.168.1.113

2023-09-13 16:01:09
IP online
192.168.1.113
```

outdoorbits, 2023-09-13:
Could you please update and check if IP repeats like above? If not - idletime should work now...? At least here it does.

outdoorbits, 2023-09-18:
Every time the IP is displayed (or tried to display) this is written to the log. The IP should not be displayed again, if it is still visible (on the display). And because idle time is calculated by the time since last change of the log files, it does not shutdown.
Unfortunately I could not fully reproduce the problem but now I have an idea, as you don't use a display, this is my start point to debug!
I wonder about the faster transfer you observed in the older version. At least transfer from storage to storage (not from camera) is optimized now and in my observations it is about three times faster (RPi 4). That's because for local backups cpu intensive tasks like compression and r/w intensive tasks like delta-transfer algorithm are disabled. For cloud services the configuration is different.
I still hope to get your feedback. If you don't mind, maybe you could compare transfer time and tell me?

outdoorbits, 2023-09-18:
Idle time problem is fixed.

chatulim, 2023-09-18:
> Idle time problem is fixed.

Confirmed. I updated, checked the "If display is activated, print IP every minute" box, set the idle time to 2 minutes, rebooted, and... the LBB powered down two minutes later. 

I think we can safely say this issue is closed.

No promises on timeframe, but I can try to run some speed tests pitting the new LBB against the older version that I am using. If the newer version is noticeably slower I will open a new issue

closed 2023-09-18 by chatulim

outdoorbits, 2023-09-18:
I'm very interested in your speed tests, please tell me, whatever they say.

outdoorbits, 2023-09-21:
This is very interesting. In your version I expect progress check should be much more read intensive because is repeats counting the files at target. But I might be wrong, not sure when this has been changed to interpret rsync output.
Could you please have a look into backup.sh?  I would like to know the arguments given to rsync...

chatulim, 2023-09-21:
> Could you please have a look into backup.sh? I would like to know the arguments given to rsync...

I'm sorry, I'm not sure exactly what you are asking for and I also don't know how to get it...

Do you want to see something in the **old** version of LBB or in the **new** (development) version?

What *exact* incantation do I have to type to get the information you desire?

outdoorbits, 2023-09-21:
Could you have a look at the old version from 2019, there must be a file names backup.sh. I expect two lines having a rsync command, one with dry-run argument. I'm interested in the other one without dry-run.

chatulim, 2023-09-21:
There are three different files in the "scripts" folder with "backup.sh" as part of their name: "camera-backup.sh", "card-backup.sh", and "internal-backup.sh"

- "camera-backup.sh" does not have the word `rsync` at all.
- "card-backup.sh" has only one line: `rsync -avh --info=progress2 --exclude "*.id" "$CARD_MOUNT_POINT"/ "$BACKUP_PATH"`
- "internal-backup.sh" has only one line: `rsync -av "$STORAGE_MOUNT_POINT"/ "$BACKUP_PATH"`

I don't know what any of this means, but I assume you do...

outdoorbits, 2023-09-21:
Oh, that's surprising to me, that's before may time. ;-)
The time lost is because processes are very static here and there are no security mechanisms for lost devices for example. Maybe saving that seconds isn't an option... ;-)