outdoorbits/little-backup-box #240: No read/write permission for subfolders inside the Backup folder

chatulim, 2023-09-21:
**Describe the bug**
When mounting the LBB via Netatalk(?) in the Finder on my Mac, the folders *inside* the "Internal" shared folder all have permissions set to **no access**.

**To Reproduce**
Steps to reproduce the behavior:
1. Plug in LBB to USB power 
2. In the Finder on a Mac, use "connect to server" to view the contents of the LBB (don't know how this would translate to other systems).
3. Select the "Internal" shared folder
4. See that all of the sub-folders have the "do not enter" badge.
5. Try to view the contents of a folder, get rejected.

**Expected behavior**
In at least one of the prior iterations of the development version of LBB, the permissions on the subfolders were set to "read/write" -- as expected. At the moment, I can manually change the permissions to be able to view the contents of the backed-up folders, but it adds an unnecessary additional step.

**Screenshots**

[image]  
[image]  
[image]

**Desktop (please complete the following information):**
 - Raspberry Pi Zero 2 W running Bullseye
 - LBB Development branch
 - Mac Mini running Mac OS 13.4 Ventura

outdoorbits, 2023-09-21:
I do not use apple devices, so I can't test this. But I just browsed the folders via filebrowser/smb without any problems.
When I login via ssh and check the permissions, I get this:
`pi@raspberrypi:~ $ ls -la /media/
total 152
drwxrwxrwx+  5 www-data www-data   4096 Sep 20 21:22 .
drwxr-xr-x  18 root     root       4096 Sep 17 09:12 ..
drwxrwxrwx+  3 www-data www-data   4096 Sep 12 20:17 backgroundimages
-rw-r--r--   1 www-data www-data    368 Sep 17 09:26 buttons.private.cfg
drwxrwxrwx+  3 www-data www-data   4096 Sep 18 15:43 internal
drwxrwxrwx   6 www-data www-data 131072 Sep 20 21:22 local_target
-rw-r--r--   1 www-data www-data   1005 Sep 17 09:26 rclone.conf`

All folders are owned by www-data and permissions are rwx to the world. Is it the same on your device?

chatulim, 2023-09-24:
> I do not use apple devices, so I can't test this. But I just browsed the folders via filebrowser/smb without any problems. When I login via ssh and check the permissions, I get this: [...]

Here's what I get with the same command:

```
pi@lbbox-ii:~ $ ls -la /media/
total 20
drwxrwxrwx+  4 www-data www-data 4096 Sep 20 21:54 .
drwxr-xr-x  18 root     root     4096 May  3 03:26 ..
drwxrwxrwx+  3 www-data www-data 4096 Sep 20 18:22 backgroundimages
-rw-r--r--   1 www-data www-data  368 Sep 20 21:39 buttons.private.cfg
drwxrwxrwx+  4 www-data www-data 4096 Sep 21 16:43 internal
-rw-r--r--   1 www-data www-data    0 Sep 20 21:39 rclone.conf

outdoorbits, 2023-09-24:
This means, permissions on file system level are correct. smbd also allows full access to user lbb. I added a samba server restart command after creating a media folder now. Because I can't reproduce the error, I can't check the effect. Any changes?
Just for understanding: You are using an apple device and wont to browse the samba shares, right? If I'm right, Netatalk isn't used to mount network paths but to share, right? It's a server, not a client. Isn't it?

chatulim, 2023-09-26:
> This means, permissions on file system level are correct. smbd also allows full access to user lbb. I added a samba server restart command after creating a media folder now. Because I can't [...]

I am not sure about the specifics of Samba vs. Netatalk, so I may be using incorrect terminology. I am using Apple devices (Mac & iPhone), and I want to browse the folders on the LBB that are shared over the network. Mostly because that is the easiest way for me to access the backed-up files on the internal micro SD card of the LBB (since I only have Apple devices, which can't read Ext4).

When I try to see the files on my Mac, I get an "incorrect permissions" message. However, when I try to do the same from an iPhone, I have no problem drilling down into the folders to see the image files.

What makes this particularly weird, is that it is still possible on the Mac for me to manually set the relevant sub-folders to "Read/Write" -- using the Mac GUI (not the terminal)... I'm not sure if this is a bug in Mac OS or some specific quirk of LBB (or Raspberry Pi, or Linux, or ???), but this is how it is. 

[image]

[image]

[image]

outdoorbits, 2023-09-26:
Very hard to say for me. As it is working on your iPhone but not on your 
mac - but ... I found an interesting site, describing how to get better 
mac compatibility:

https://wiki.samba.org/index.php/Configure_Samba_to_Work_Better_with_Mac_OS_X

As I can not test it, could you figure out the changes needed in 
/etc/samba/smb.conf?
You can test it by editing it:

sudo nano /etc/samba/smb.conf

and

sudo service smbd restart

As soon as you find it out, I'll update lbb.

Thank you!

chatulim, 2023-09-26:
> https://wiki.samba.org/index.php/Configure_Samba_to_Work_Better_with_Mac_OS_X As I can not test it, could you figure out the changes needed in /etc/samba/smb.conf? You can test it by editing it: sudo nano /etc/samba/smb.conf and sudo service smbd restart 

Just to clarify: I should make the edits to the samba file on **LBB** (not on my Mac). Please confirm.

chatulim, 2023-09-26:
> Would you please update development version and check the function again? I just added the samba config part.

I reverted to the old samba config to make sure the problem came back (**it did**). Then I updated to the latest development version to confirm that the problem was gone. **It was**.

So that's it for me. As far as my testing goes, you are ready to ship.

;-)

closed 2023-09-26 by chatulim