outdoorbits/little-backup-box #241: Can't back up to cloud services

CossieChch, 2023-09-22:
Hi

I have got lbb working successfully, backing up from usb source and camera source to both internal and external usb - thanks!  I have set up two cloud services (Google Drive and Google Photos) as detailed in the "04b. Cloud services"  link.  However, when I choose these options from lbb, nothing appears to happen, with nothing written to the log.  Am I missing a setup step?  I can successfully write to these cloud services using rclone from my laptop, which is how I generated the rclone.conf entry (copying from the file generated on the laptop) 

Cheers

outdoorbits, 2023-09-22:
Are you using development version? There is a bug I'll fix maybe Sunday. Backup works but you get an error lost cloud device.

outdoorbits, 2023-09-23:
There was bug in mount process. For me it is working now (development branch). Could you please test?

closed 2023-10-01 by outdoorbits