outdoorbits/little-backup-box #245: Backup not starting [dev branch; specific Hardware ]

samuel-puschacher, 2023-10-05:
I am currently Testing the developement Branch with the following Raspberry CM4 based Hardware
https://www.waveshare.com/cm4-nvme-nas-box.htm

It seems to me as the backup does not start /devices are not detected correctly:
`Little Backup Box
IMAGE=/var/www/little-backup-box/little-backup-box.bmp

2023-10-04 23:47:02
online, IP:
x.x.x.x
x.x.x.x
x:x:x:x:x:x

2023-10-04 23:55:31
USB storage 
 > USB storage 

2023-10-04 23:55:31
Source: usb 

2023-10-04 23:55:31
Target: usb 

2023-10-04 23:55:31
mount device: usb, target; WaitForDevice=True; DeviceIdentifierPresetThis=; DeviceIdentifierPresetOther=

2023-10-04 23:55:31
MOUNTED?: '/media/local_target' = False

2023-10-04 23:55:31
umount /media/local_target: Not mounted.

2023-10-04 23:55:31
Ready
Insert target

2023-10-04 23:55:31
pre mount usb, target (device list changed)
PATH="/dev/sda" MOUNTPOINT="" UUID="" FSTYPE=""
PATH="/dev/sdb" MOUNTPOINT="" UUID="" FSTYPE=""
`
after that there is no Progress

The Devices are Recogniced in the System:

`df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  4.5G  9.0G  33% /
devtmpfs        1.7G     0  1.7G   0% /dev
tmpfs           1.9G     0  1.9G   0% /dev/shm
tmpfs           759M  3.0M  756M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  255M   60M  196M  24% /boot
tmpfs           380M   20K  380M   1% /run/user/1000
/dev/sda1       120G   21G   99G  18% /media/lbb/3233-3230
/dev/sdb1       3.7T  100G  3.6T   3% /media/lbb/Extreme Pro
`
The USB-Devices:
`lsusb
Bus 003 Device 002: ID 0c76:161f JMTek, LLC. USB PnP Audio Device
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0781:55af SanDisk Corp. Extreme Pro
Bus 002 Device 004: ID 11b0:3306 ATECH FLASH TECHNOLOGY UHS-II SD Reader
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

The Devices are connected over the VL805 USB 3 Hub
[image]

outdoorbits, 2023-10-05:
If I'm right, the device you want to use by LBB is already mounted. Lbb looks for not mounted devices in its process. If you need to use a always mounted partition as target, this may be possible replacing/media/internal by a symlink to your target path.
Does this describe/solve your problem?

samuel-puschacher, 2023-10-05:
Yes this is my Plan for the internal M.2 SSD, but I have not Populated it yet, at the moment I am testing with two USB devices.
What is interesting, in the Log there is no UUID and Mountpoint found, but when executing lsblk there they are. Maybe a Issue with the grep in the following Line: 
USB_DEVICE="$(sudo lsblk -p -P -o PATH,MOUNTPOINT,UUID,FSTYPE | grep "^PATH=\"/dev/${DEVICE}\"")"

https://github.com/outdoorbits/little-backup-box/blob/e846cbdd13d465ef0b049b490094a2b01d796d66/scripts/tools-fsck.sh#L25

outdoorbits, 2023-10-05:
But the your test devices are mounted before you start the backup, right? That means, LBB won't interpret them as available devices.
USB devices can't be used by LBB if they are already mounted. The only exception I see could be a workaround via /media/internal.

In your log the LBB script waits until a device is connected.

samuel-puschacher, 2023-10-05:
Hi you are right, was testing with Main branch on different hardware before where I had both devices connected at the beginning.

closed 2023-10-05 by samuel-puschacher