outdoorbits/little-backup-box #249: Problem or Operator Failure

tpalmer0127, 2023-10-16:
I was able to load the software on a pi zero w with the zero 4 u card attached
I used the pi imager, updated online once loaded and then used the curl command to load lbb

I'm using a Seagate USB hard drive, powered USB hub, and SD card reader
I can access the system with the Web UI
Stop box and Text - red
I select the Source - USB Storage - box red outline
and Target USB  - box dark outline
The load scroll bar on the phone moves about 1/4" and then disappears and UI screen seems to re-load
At one point I think I saw the STOP box change color, but not today

I've tried booting with the SD & USB HD attached or attached after re-boot with no difference

I have the file manage open on the pi HDMI connection and see no movement of files

Folders and Files on the SD
NIKON Z 7
  DCIM
    162NCZ_7 - contains 6 .NEF files
    Test - contains 6 .JPG files

In the zero file manager I see
/media/tdp/NIKON Z 7/DCIM/162NCZ_7
/media/tdp/NIKON Z 7/DCIM/Test
/media/tdp/Seagate Expansion Drive/  ... various folders

UI is on iPhone 12 Pro
ios 16.6.1
Browser Safari

Desktop is HDMI monitor, Logitech wireless USB Keyboard, and mouse connected to pi

Any suggestions would be appreciated.
Thanks
Terry

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

outdoorbits, 2023-10-16:
I'm not sure I understood your situation fully. What's the load scroll bar? And what does the quarter inch mean?
LBB UI reloads just the monitor area frequently.
What surface are you using on HDMI? Are you using raspbian lite?
Could you set log level to 3 and publish your log file?
Did you configure mail? What's the result?

tpalmer0127, 2023-10-16:
outdoorbits
Thanks for the reply

Scroll bar is the load status bar that moves left to right across the bottom of my phone to show the status of the web page load process.  I would say on a normal web page load, it "scrolls" fully left to right at the page is displayed.
When it stops partway, that usually indicates the load of the page is stuck or failed
For example, when I'm on the UI, goto Menu and select setting, the "scroll" moves fully left to right and the Settings Page appears

Surface ?  If you mean screen and keyboard directly connected to the pi, I'm using a standard HDMI monitor connected the the pi HDMI port and a Logitech wireless USB mouse / Keyboard plugged into a port on the zero 4 usb board.  If you mean the UI device, it is an iphone connected to the pi wireless ip address

Not using raspbian lite, using the full version

Did not configure mail unless it was part of the "curl" procedure.  I did not see that mentioned in the instructions.

I did try using various web browsers on the pi, but gave up and did a fresh os install which is where it is now

I can try the log again.  
Tried Log level 3, Log sync-protocol checked, Save
little-backup-box.log is empty

Terry

tpalmer0127, 2023-10-16:
Regarding the log
I goto the menu
Select behavior
Set Log level 3
Check Log sync-protocol
Save
Menu
Select Backup
Menu
Select Behavior
Log level is set to 1 and log sync is unchecked

outdoorbits, 2023-10-17:
O.k., don't mind the load bar, some pages of the web UI have a refresh function, so it is reloading high frequent. But you get the full page in your browser, right?
In my question about the surface I wanted to know if you are using a graphical interface as you do. In this case there could be interactions between the usb assistance of your system and lbb. This is why you should use the light version. Please try at least everything first using the light version. When you managed to run lbb there and want to try lbb on the full version anymore, please post your experiences.
Mail configuration can be done in settings.
If your changes in settings are not saved as you describe, I expect a specific problem about your installation. I just tried this here without any problems. I just started a new installation to check that again.
Anyway , the log of course needs action to get populated. Did you try to start a backup and check the log?

To get started I would ask you to reinstall lbb on a not modified raspberry pi os lite on sd.

tpalmer0127, 2023-10-17:
Thanks for the suggestion.  Iwas thinking the load bar was probably as you describe.  I'll try the lite os
BTW, I see the pi connect to my local network when it boots.

Yes, the HDMI out of pi is connected to a monitor with kybd and mouse so there is a graphical ui connected directly to the pi

Do I need to configure mail ?

T

outdoorbits, 2023-10-17:
There is no need to configure mail, except if you want to get a mail report about backups.

tpalmer0127, 2023-10-17:
After several false starts, I'm using the lite os, 32 bit
Started the wifi and am loading  lbb

A couple of quick questions while lbb loads.
Is it possible to load a gui ?
What about a browser ?
Thanks
T

outdoorbits, 2023-10-17:
Of course, it's all described at https://github.com/outdoorbits/little-backup-box
There is an issue about comitup witch allows multiple ways of wifi connections - that's broken actually. I'm working on a solution.

tpalmer0127, 2023-10-18:
Good news.  It appears to be working.  the UI on the phone is  now a black background rather than white.  Not sure of the importance of that. The log monitor shows activity, and when I look at the USB HD, the files have been copied

Thanks for helping me through that.  I had a lot of false starts and re-imaging to get to this point.

I'm not finding the information you mentioned in your previous e-mail on how to install a gui and browser on the surface attached to the pi  I'd like to be able to get to a file manager on the surface.  And, is it possible to see a file listing from the web UI ?  I tried the Mount function but that doesn't seem to allow a directory type view.

I tried raspi-config and changing the boot to GUI, ( requiring lightdm to be installed) but the system loads part way and stops with a blank screen

Terry

outdoorbits, 2023-10-18:
There is no local GUI to connect a monitor directly. The only GUI is the web interface. To view files on your disks, just mount them (using Tools) and browse them using Files. Both (Tools and Files) are menu items in the upper part of the web page.
As I described before, I would strictly not recommend to use a graphical system. The lite version does the job. Also any changes on the system (raspi-config) could cause unexpected behavior of lbb.

outdoorbits, 2023-10-18:
Select the bullseye version!

tpalmer0127, 2023-10-20:
Good evening,
I kind of gave up on the pi zero and put the card in a faster version (4 usb ports, hdmi-a port and ether port)  Much more responsive !!  Trying to set up e-mail and I'm getting Error sending maill:  cannot access local variable 'server' wher it is not associated with a value  
My lbb e-mail settings
check send mails
check html format
SMTP  smtp.mail.yahoo.com
port  465
Connection  SSL
username - my yahoo e-mail
pw - my yahoo pw
sender - my yahoo e-mail
recipient - my yahoo e-mail
check send test mail

What's the bullseye version ?

tpalmer0127, 2023-10-20:
OK, I see the bullseye version on the imager

outdoorbits, 2023-10-21:
I think to switch to a faster pi is a very good decision. If you don't 
mind the money, the RPi 5 (announced for 2023/10/23) is expected to 
bring much more advantages. Maybe most important will be a better power 
supply for USB ports. But also it is faster processing and transferring.

About the mail problem:
This message means, there was a problem about your mail configuration 
and your connection could not be established. If you update lbb (by web 
ui - settings) you'll get a little improved information about the 
problem now. Would you maybe try to setup STARTTLS?

You maybe will get more than one error message in ylor log monitor. What 
do they say?

Do you use special characters in your password? Handling special 
characters in this context is tricky and maybe I didn't escape everyone.

Did you check your settings using another mail client like thunderbird?

tpalmer0127, 2023-10-21:
Probably worth the wait for the 5
My PW uses _ and &

I tried a gmail email account with no special PW characters
This time I'm getting Error sending HTML mail:  (530, b'5.7.0  Authentication Required

tpalmer0127, 2023-10-21:
Well,  I had to delete 2 factor authentication., 

Now getting Mail error while defining SSL smtp server  
Username and PW not accepted.  
Then
Error sending plain text mail
Authentication required

I've verified and re-entered info into lbb settings ???
but still get Authentication error

Also got
Error sending plain text mail:  'ascii' codec can't encode character '/uf029' in position 58900 : ordinal not in range(128)

now getting
Mail error while defining SSL smtp server:  Connection unexpectedly closed
Error sending HTML mail: please run connect() first
Mail error while disconnecting from smtp server: please run connect() first
etc.

I seem to be getting different errors with the same e-mail setup ?

tpalmer0127, 2023-10-21:
I've tried two
smtp.gmail.com
465
SSL
user my gmail e-mail
PW my gmail pw with no special char
sender my gmail e-mail
recipient my yahoo e-mail

and
smtp.mail.gmail.com
similar to above except all e-mails are my yahoo e-mail, and my yahoo pw has special char.

outdoorbits, 2023-10-21:
This weekend I won't find the time to check this. But I'll do some tests next week and try to catch the error.

outdoorbits, 2023-10-21:
... But did you try to use your settings using another mail client software like thunderbird?

tpalmer0127, 2023-10-21:
Yes, setup T-bird their default with no problem
The only difference I see is lbb offers SSL, T-bird sets up SSL/TLS
and T-bird shows authentication method  OAuth2 where lbb has no field

tpalmer0127, 2023-10-21:
I tried another e-mail, an office365, similar problems.  Please see attached image
[image]

outdoorbits, 2023-10-21:
A quick Google check told me to check every combination of ports 465 / 587 and SSL / STARTTLS.
Does one of these 4 options work?

tpalmer0127, 2023-10-22:
I'll do an exhaustive check of the combinations

outdoorbits, 2023-10-22:
Your google account uses "2-Step-Verification", right? In this case you need a app password for lbb. Login into google, got to "Manage your google account", "Security", "2-Step Verification", "App passwords". Here you have to create an app password. Use this (without the spaces) in your mail settings: username and sender=gmail adress, ssl and port 465.
It ran without any problems.

tpalmer0127, 2023-10-22:
I disabled 2 stop in gmail, but i'll try this
T

outdoorbits, 2023-10-24:
Have you been successful? I tested again some providers but if trouble persists and it's related to lbb, I need to know.

tpalmer0127, 2023-10-24:
I've run out of time.  Heading from PA to FL.  Should be able to get back to this next week (11/1)
Do  you have a URL for the above procedure re app password.
I had deleted 2 step verification, so not sure if this applies
Thanks
Terry

outdoorbits, 2023-10-24:
I can't give a direct link but after login at goggle just follow the steps mentioned in my last mail. Here you'll find all the security options.

outdoorbits, 2023-11-05:
Could you fix the problem? If yes, I would like to close this issue.

tpalmer0127, 2023-11-05:
I haven't had time to get back to this.  We took a week to get to Florida and another to unpack.  I should be able to unpack that box this week and try your suggestions.I'll let you know either way.Terry

tpalmer0127, 2023-11-10:
Let's close this issue.  We're packing for 2 weeks in Hawaii and it doesn't look like I'll get back to this before early December.If I still have a problem, I'll create another support request.Again, Thanks for your helpT

closed 2023-11-10 by outdoorbits