outdoorbits/little-backup-box #259: Instructions for defining source and target devices (and paths)

gadgetmies, 2023-12-11:
**Is your feature request related to a problem? Please describe.**
Not really a problem per se, but I cannot find any instructions on how to define the source and target devices. My target device has multiple partitions (e.g. boot and root for the RPi) and thus by default the script selects the wrong source and target.

**Describe the solution you'd like**
I would like to at least define the source and target block devices (i.e. e.g. /dev/sdd4). Currently the UI and the backup.py script seem to automatically select the devices, which for my case does not work. Perhaps the configuration UI could just have an option for defining the devices using a simple input (a drop down would be even better). 

**Describe alternatives you've considered**
I noticed that the backup.py script has a DeviceIdentifierPresetSource and DeviceIdentifierPresetTarget options that sound like could be the thing I need. However I was unable to find instructions for command line use and documentation on the arguments. Adding some documentation to the wiki on how to use the script from command line would be nice. Even better would be that the script would print out instructions for use when it is run without proper arguments.

outdoorbits, 2023-12-11:
I could imagine to add a device preselection to to web UI - just like the other modifications. Would this do the job for you?

gadgetmies, 2023-12-11:
Perhaps, if you could point me in the right direction. Perhaps documenting the command line arguments to the script could be the first step so that I can verify I understand the code correctly. Not that confident with Python, so assistance would be appreciated!

outdoorbits, 2023-12-11:
If you work on the command line, you could use rsync directly. You can 
find all the arguments at the end of backup.py,below "if __name__ == 
"__main__":".

outdoorbits, 2023-12-12:
I just added the option to define partitions in web UI (see modifications). Does this help?

gadgetmies, 2023-12-12:
Wow! Seems like just what I'd need! Thanks!

Cannot test the functionality right now however, as I somehow busted my RPi SD card. I'll try to find time to test in the upcoming days.

closed 2023-12-18 by outdoorbits