outdoorbits/little-backup-box #265: Raspberry Pi 5: GPIO functions (display and buttons) do not work

outdoorbits, 2023-12-29:
There are major hardware changes and essential components are not adapted yet. Lbb does not yet support display and buttons on Raspberry Pi 5.

outdoorbits, 2023-12-31:
First part is fixed: Buttons work now - but the display still remains black on Raspberry Pi 5.

PPrecht, 2024-01-03:
Hello, I received my Raspberry Pi5 today and installed Pi OS Bookworm along with Little Backup box and included Comitup. After I entered the access data for my Fritzbox router on the Comitup page, the Pi does not provide a hotspot when the Fritzbox is nearby. Even deleting the access data in the LBB menu (Comitup) does not bring about any problems?

outdoorbits, 2024-01-04:
The first part is clear, as soon as there is a connection to a WiFi, Comitup stops it's hotspot. If you still need a hotspot, you can use a usb WiFi dongle, in that case it will hold both services.
The second part is not clear. I had a similar situation. I even got the message Comitup is starting the hotspot on display but it didn't come up. After reboot it worked. Hoping for the upcoming version of Comitup...

PPrecht, 2024-01-04:
Hello from germany,

 

i had like to  make my Pi5 powered LBB equipped with a OLED Display I2C SSD1306.

At Github i could read that at this time is noch working with bookworm.

 

Any chance that i will work in the future?

 

Thanks

Peter

outdoorbits, 2024-01-04:
I hope so, maybe soon. I'm just testing options.

outdoorbits, 2024-01-13:
It is working! There is a temporary (?) workaround until the patch will maybe be included into luma.core. Just update you lbb.

closed 2024-01-13 by outdoorbits