outdoorbits/little-backup-box #268: Mail does not work when "user" is not an e-mail address

FG1965, 2024-01-09:
Hello
I'm trying to use the email feature with my standard e-mail setup. 
In mosts scenarios, the required SMTP-data are "e-mail address as USER" and the password.

However, in my case the "user" in the SMTP-Setup is different from the e-mail address, "user" is something like "wp123-report".
This is how the system is setup from my provider.

LBB supports to enter a dedicated user name. Great!

However, I always receive the follwing error:

> 2024-01-09 11:25:40
> Mail error while defining STARTTLS smtp server: local variable 'context' referenced before assignment
> 
> 2024-01-09 11:25:40
> Error sending plain text mail: (501, b'<wp123-report>: sender address must contain a domain', 'wp123-report')

I tried with
- bookworm or bullseye
- STARTTLS or SSL
- port 25 or 587
- HTML or plain
- passwords with and without special characters

Hence, I assume this bevaviour is a bug.

As a workaround I've created a new gmail-address (where the user is equal to the e-mail) and everything works fine.

Best regards
Frank

outdoorbits, 2024-01-09:
You are right and thank you very much for your intensive research. I had a bug in lib_mail.py and fixed it. Could you please check if the problem is solved by this fix?

FG1965, 2024-01-09:
Thanks for the fast response!
I did an update (clicking on the menu item UPDATE) and configured the server.

When sending with **port 25 and STARTTLS** the result is
> 2024-01-09 13:37:21
> Error sending HTML mail: (501, b'<wp123-report>: sender address must contain a domain', 'wp123-report')

When sending with **port 25 and SSL** the result is
> 2024-01-09 13:40:25
> Mail error while defining SSL smtp server: [SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:1123)
> 
> 2024-01-09 13:40:25
> Error sending HTML mail: local variable 'server' referenced before assignment
> 
> 2024-01-09 13:40:25
> Mail error while disconnecting from smtp server: local variable 'server' referenced before assignment
> 

The provider says I should use **port 587 and STARTSSL** which gives again

> 2024-01-09 13:45:46
> Error sending HTML mail: (501, b'<wp123-report>: sender address must contain a domain', 'wp123-report')
> 

Either the error is not yet fixed or I did the wrong update-procedure.
Sorry !

outdoorbits, 2024-01-09:
The update fixed the context variable error. I'll investigate...

FG1965, 2024-01-09:
If it helps: I can setup a temporary e-mail on my server for testing purposes.

Is there a way to send a private message with the setup data ?

outdoorbits, 2024-01-09:
Great! Mail addresses are filtered, so please have s look at the header of any .py file in scripts folder of the repository.

FG1965, 2024-01-09:
The problem has already been solved with the newest update!!!
Thanks a lot for the fast response !

closed 2024-01-09 by FG1965