outdoorbits/little-backup-box #270: Unable to login after global password set

1bryanw, 2024-01-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

outdoorbits, 2024-01-15:
Does your password contain special characters?

1bryanw, 2024-01-15:
No special characters only letters and numbers. total 8

outdoorbits, 2024-01-15:
Witch login fails? Web UI and SSH?

1bryanw, 2024-01-15:
Web UI fails.  I just reinstalled so ready for any possible troubleshooting. I'm sure If i try to change global password I can recreate if needed.

outdoorbits, 2024-01-15:
The user name for the web UI is "lbb".

1bryanw, 2024-01-15:
Ok I'll try that after changing global password.  I'll let you know once I give it try in about an hour.
Thanks

1bryanw, 2024-01-15:
Ok that was it. I guess I missed that part, but its on documentation so my fault for not catching that.

Thank you so much for quick reply.
Bryan

closed 2024-01-15 by 1bryanw