outdoorbits/little-backup-box #276: Unable to backup to Samsung T7 Touch

ltonn02, 2024-02-23:
Whenever I try to backup to my Samsung T7 Touch (encryption disabled) it tries to backup itself to itself.

This is the log-file. It seems like there are two filesystems on that drive. When you use the encryption feature of that ssd, you are prompted with a read-only filesystem. Maybe this is read, even though encryption is disabled? Is there any way to work around that? Maybe prohibit it from mounting the vfat fs? Thanks in advance.

Little Backup Box
2024-02-23 17:20:19
IMAGE=/var/www/little-backup-box/little-backup-box.bmp

2024-02-23 17:20:20
Comitup:
HOTSPOT

2024-02-23 17:20:20
USB storage 
 > USB storage 

2024-02-23 17:20:20
Source: usb 

2024-02-23 17:20:20
Target: usb 

2024-02-23 17:20:21
mount device: usb, target; WaitForDevice=True; DeviceIdentifierPresetThis=; DeviceIdentifierPresetOther=

2024-02-23 17:20:21
MOUNTED?: '/media/local_target' = False

2024-02-23 17:20:21
umount /media/local_target: Not mounted.

2024-02-23 17:20:21
Comitup:
CONNECTING

2024-02-23 17:20:21
Ready
Insert target

2024-02-23 17:20:30
Comitup:
CONNECTED

2024-02-23 17:24:33
pre mount usb, target (device list changed)
PATH="/dev/sda" MOUNTPOINT="" UUID="" FSTYPE=""

2024-02-23 17:24:34
pre mount usb, target (device list changed)
PATH="/dev/sda" MOUNTPOINT="" UUID="" FSTYPE=""
PATH="/dev/sda1" MOUNTPOINT="" UUID="67E3-17ED" FSTYPE="vfat"
PATH="/dev/sda2" MOUNTPOINT="" UUID="65D7-BE8B" FSTYPE="exfat"

2024-02-23 17:24:34
DeviceChosenIdentifier='--uuid 67E3-17ED' ('/dev/sda1') prepared to mount at '/media/local_target', vfat

2024-02-23 17:24:34
MOUNTED?: '/media/local_target' = False

2024-02-23 17:24:34
mounting:
USB

2024-02-23 17:24:34
mount --uuid 67E3-17ED /media/local_target -o uid=33,gid=33,umask=0

2024-02-23 17:24:34
mounted USB '--uuid 67E3-17ED' at '/media/local_target': Msg.='$'

2024-02-23 17:24:34
MOUNTED?: '/media/local_target' = True

2024-02-23 17:24:35
Target o.k.
size: 202M
used: 202M
free: 5k
FS: vfat
 100.0% >>>>>>>>>>>>>>>>>>>>

2024-02-23 17:24:35
mount USB device: --uuid 67E3-17ED

2024-02-23 17:24:35
MOUNTED?: '/media/local_target' = True

2024-02-23 17:24:35
mount device: usb, source; WaitForDevice=True; DeviceIdentifierPresetThis=; DeviceIdentifierPresetOther=

2024-02-23 17:24:35
MOUNTED?: '/media/local_source' = False

2024-02-23 17:24:35
umount /media/local_source: Not mounted.

2024-02-23 17:24:35
Ready
Insert sources

2024-02-23 17:24:35
pre mount usb, source (device list changed)
PATH="/dev/sda" MOUNTPOINT="" UUID="" FSTYPE=""
PATH="/dev/sda2" MOUNTPOINT="" UUID="65D7-BE8B" FSTYPE="exfat"

2024-02-23 17:24:35
DeviceChosenIdentifier='--uuid 65D7-BE8B' ('/dev/sda2') prepared to mount at '/media/local_source', exfat

2024-02-23 17:24:35
MOUNTED?: '/media/local_source' = False

2024-02-23 17:24:36
mounting:
USB

2024-02-23 17:24:36
mount --uuid 65D7-BE8B /media/local_source -o uid=33,gid=33,umask=0

2024-02-23 17:24:37
mounted USB '--uuid 65D7-BE8B' at '/media/local_source': Msg.='$'

2024-02-23 17:24:37
MOUNTED?: '/media/local_source' = True

2024-02-23 17:24:37
Source o.k.
size: 977G
used: 37G
free: 939G
FS: exfat
 3.8% ____________________

2024-02-23 17:24:37
mount USB device: --uuid 65D7-BE8B

2024-02-23 17:24:37
MOUNTED?: '/media/local_source' = True

2024-02-23 17:24:37
Backup from usb: 

2024-02-23 17:24:37
MOUNTED?: '/media/local_target' = True

2024-02-23 17:24:37
Lost target device usb? False

2024-02-23 17:24:37
MOUNTED?: '/media/local_source' = True

2024-02-23 17:24:37
Lost source device usb? False

2024-02-23 17:24:37
Working...

2024-02-23 17:24:37
rsync -avh --info=FLIST0,PROGRESS2 --stats --no-owner --no-group --no-perms --mkpath --min-size=1 --exclude *.id --exclude images.db --no-compress --whole-file --dry-run /media/local_source/ /media/local_target/lbb_20240223171408-70967

2024-02-23 17:24:37
grep Number of regular files transferred

2024-02-23 17:24:45
Files to sync from folder '': 15234

2024-02-23 17:24:45
Files to sync before backup: 15234

2024-02-23 17:24:45
USB storages 
 > USB storage 
0 of 15234
Time: ?
comparing

2024-02-23 17:24:45
rsync -avh --info=FLIST0,PROGRESS2 --stats --no-owner --no-group --no-perms --mkpath --min-size=1 --exclude *.id --exclude images.db --no-compress --whole-file /media/local_source/ /media/local_target/lbb_20240223171408-70967

2024-02-23 17:24:45
Files missing: 15234 files not synced.

2024-02-23 17:24:45
Files missing:
>lsblk -p -P -o PATH,MOUNTPOINT,UUID,FSTYPE
PATH="/dev/sda" MOUNTPOINT="" UUID="" FSTYPE=""
PATH="/dev/sda1" MOUNTPOINT="/media/local_target" UUID="67E3-17ED" FSTYPE="vfat"
PATH="/dev/sda2" MOUNTPOINT="/media/local_source" UUID="65D7-BE8B" FSTYPE="exfat"
PATH="/dev/mmcblk0" MOUNTPOINT="" UUID="" FSTYPE=""
PATH="/dev/mmcblk0p1" MOUNTPOINT="/boot/firmware" UUID="5DF9-E225" FSTYPE="vfat"
PATH="/dev/mmcblk0p2" MOUNTPOINT="/" UUID="3b614a3f-4a65-4480-876a-8a998e01ac9b" FSTYPE="ext4"

2024-02-23 17:24:45
No abnormal conditions have been detected.

2024-02-23 17:24:45
Exception: Err. in file I/O, Error code '11'

2024-02-23 17:24:45
No abnormal conditions have been detected.

2024-02-23 17:24:45
MOUNTED?: '/media/local_target' = True

2024-02-23 17:24:45
Lost target device usb? False

2024-02-23 17:24:45
MOUNTED?: '/media/local_source' = True

2024-02-23 17:24:45
Lost source device usb? False

2024-02-23 17:24:46
Try backup 2 of 5

outdoorbits, 2024-02-23:
This is strange, it shouldn't mount two partitions of the same drive like sdaX. This looks like a bug. I think I'll find some time Monday or Tuesday to fix this. But this wouldn't help you much as the normal behaviour would be too use the first partition and if I got you right, you need to use the second.
Did you try to use the web UI to start backup and define the partitions to use in the modifications section?

ltonn02, 2024-02-24:
Removing `'vfat'` from the supported file types in `/var/www/little-backup-box/scripts/lib_storage.py` fixed it for me.
Maybe this can be toggled from the settings in the future?

For future reference, it should look something like this:
Line 47: `FS_Types_supported = ['ext2','ext3','ext4','fat','exfat','ntfs','hfs','hfsplus']`

I couldn't define the partitions manually because it automatically starts backing up as soon as I connect it.

ltonn02, 2024-02-24:
I'm also not 100% happy with this solution... Maybe a user changeable setting to specify a minimum partition size? Or just auto selecting the largest partition instead of the first one?

outdoorbits, 2024-02-24:
Defining a minimal size is my idea also.

ltonn02, 2024-02-26:
Unfortunately no. When I connect my SSD, everything seems to be working just fine (it mounts the correct partition), but as soon as I connect my SD card it seems to get stuck in some kind of loop... you'll find the logs attached below.
[little-backup-box.log](https://github.com/outdoorbits/little-backup-box/files/14404489/little-backup-box.log)

outdoorbits, 2024-02-26:
Could you please try again? It looks like your sd partition has no uuid identifier, this is common in sony devices. I think it's fixed now. Is it?

outdoorbits, 2024-02-26:
It tries to mount sdb1:
`mount sdb1 /media/local_source -o uid=33,gid=33,umask=0`

Can you check the filesystem, is it ok?
Could you try to execute this:
`sudo mkdir /media/local_source
mount sdb1 /media/local_source -o uid=33,gid=33,umask=0`

Is there an error message?

ltonn02, 2024-02-26:
`sudo mkdir /media/local_source mount sda1 /media/local_source -o uid=33,gid=33,umask=0`
`mkdir: invalid option -- 'o'`
`Try 'mkdir --help' for more information.`

are you sure about the command? Filesystem seems to be working... can read/write files from pc and camera.

ltonn02, 2024-02-26:
using those commands I was able to mount the sd card...

`sudo mkdir /media/local_source`
`sudo mount /dev/sda1 /media/local_source -o uid=33,gid=33,umask=0`

outdoorbits, 2024-02-26:
You pointed the error. I forgot the /dev/ before sdX. Is it working now? My tests where successful.

ltonn02, 2024-02-26:
Works perfectly. Thank you!

closed 2024-02-26 by ltonn02