outdoorbits/little-backup-box #286: Update freezes at "Setting Up Comitup (1.41-1) ..."

macmcb, 2024-04-01:
**Describe the bug**
Update process via web UI appears to hang (~20 minutes without movement). "Setting up Comitup (1.41-1)" is the last line shown in the on-screen log file.

I read[ this troubleshooting advice](https://github.com/outdoorbits/little-backup-box/wiki/07.-Troubleshooting#update), opened a second web ui tab and hit Update again. The second log window shows the same "Setting Up Comitup..." message and no movement.

LBB is running on Pi 4B with Bookworm (Lite)

**To Reproduce**
1. In web UI: Settings > Update
2. Wait and wait...

**Expected behavior**
Should display "Finished"

**Desktop (please complete the following information):**
 - Windows 10, Firefox 124.0.1

outdoorbits, 2024-04-01:
To repair samba for windows shares I had to exchange PHP. This seems broke the update. Would you please update from command line? Just as you started your installation. I'm already working on the issue.

macmcb, 2024-04-01:
Update from command line worked. Samba shared folders now working from Windows. Amazing! Thank you very much!

closed 2024-04-02 by outdoorbits