outdoorbits/little-backup-box #290: Question About Transfer Speed

GypsyAcidQueen, 2024-04-09:
Hi Dmitri and team! First of all wanted to say thank you so much for this amazing project. I have been researching for days how to hack something like this together myself and through a random Reddit post stumbled on this repo. It's far better than anything I could have done myself. 

I just wanted to ask if the project supports usb 3.0 transfer speeds on the pi 4 and 5 as I couldn't find any information on here or the site. My use case will be transferring SD card footage (hundreds of gigabytes) to an external ssd. My v90 cards are capable of 400 mb/s read, will little backup box be able to support that?

Many thanks again!

outdoorbits, 2024-04-09:
This does less depend on the software. Of course, the USB 3.0 is working 
but it is limited to the raspberry pi. pi 4 and 5 have USB 3.0, pi 5 has 
the better controller and USB to USB transfer is faster compared to the 
pi 4. But anyway, the pi never reaches what to expect from USB 3.0. In 
my experience rates close to 200 MB/s are possible. Usually I get stable 
rates from 80 to 150 MB/s. This may not only be limited by pi's USB 3.0 
but also by the CPU. Backup is performed by rsync and this means it 
checks differences between source and target - this takes some 
performance but increases security.

GypsyAcidQueen, 2024-04-09:
Understood, thank you for your reply Outdoorbits! I think these speeds are definitely workable for me, is it the case that a single continuous transfer will vary from 80-150, or that you see a range of stable speeds between 80 and 150 depending on the cards and drives etc used?

outdoorbits, 2024-04-09:
It depends a lot on the drives. From USB 3.0 to internal NVMe SSD I have 
seen much higher rates. During transfer you get speed information based 
on the last transferred file before the message is created. The 
efficiency seems to depend on the file size, maybe on other factors 
also. This means, the speed you see is not the average but the value of 
just the last file. That's why you see fluctuations.