outdoorbits/little-backup-box #298: Try starting the backup via buttons USB->USB fails. After confirming USB-USB backup - backup is just stopped but no new backup cycle is started

SigmaFred, 2024-04-24:
I am trying to use LBB with a Raspi 3b+ (also verified with Raspi 5) with a 0.96 inch display and 4 buttons for the menu
Installation works fine and I can use the buttons and display to scroll through the menu.
Initial automated backup seems to work - but I want to start my backups manually, as I want to backup in 2 cycles (2 separate independent targets) 

All looks good, but when trying to initiate a backup via the menu 
+backup => USB > USB  => question "Confirm ..." => Yes ... 
LBB shows "Backup stopped" ... but nothing else happens ... no request to insert target or similar
Am I mistaken in my expectation that the backup cycle is initiated (reinitiated) via this menu entry?

outdoorbits, 2024-04-24:
Bug was simpler than expected. Please test again.

SigmaFred, 2024-04-24:
Thank you for the bugfix - works like a charm now :-)

closed 2024-04-24 by SigmaFred