outdoorbits/little-backup-box #301: Installation failing on raspberry pi 5 8gb

chrisfoster97, 2024-06-18:
**Describe the bug**
I have tried installing it several times across different sd cards, every time I get to the end it either fails and freezes after doing comitup, or when I didn't do comitup it would still get to the end, say it was rebooting and then after a few seconds the light on the pi would go red

**To Reproduce**
Steps to reproduce the behavior:
Followed instructions shown in the guide. Select mode 2 and yes for install comitup. It goes through the installation steps and the light goes red in the middle of installation, the last step it was on was "update-initramfs: Generating /boot/initrd.img-6.6.20+rpt-rpi-2712".

Reinstalled os on SD card, this time 32 bit lite instead of 64 bit lite, command line says setup was finished. I was briefly able to access the web gui. On every other reboot attempt the green light would be on with a blinking red light, then it would go full red after 5-15 seconds

[putty log.txt](https://github.com/user-attachments/files/15879664/putty.log.txt)

outdoorbits, 2024-06-18:
There is a report about an installation issue in comitup: 
https://github.com/davesteele/comitup/issues/245

I'm a bit in a hurry this days. Could you please try to install comitup 
on a fresh bookworm? It looks like Dave Steele has simplified the 
installation process.

Does this work ? https://davesteele.github.io/comitup/ppa.html (follow 
"The Better Way") If yes, I would rework the lbb installation.

Thank you!

outdoorbits, 2024-06-18:
I did minor changes in comitup installation, but nothing I see as relevant for function. Anyway, I just installed successfully lbb & comitup.
Maybe the problem is different. Did you configure your wifi (by Raspberry Pi Imager or using the comitup portal page)? If you don't have an extra wifi dongle, IP 10.41.0.1 will not be valid after connecting to the wifi. You get another IP. This you can see on the display or you get a mail containing the link(s).

closed 2024-06-21 by outdoorbits