outdoorbits/little-backup-box #314: Comitup Url 404 Error

rolkow, 2024-07-13:
Hi
I tried to install LBB on a Raspi 4, but there are for the comitup-installation  always errors 404 - File not found -  in the install-error.log.

When I try to access the file via browser with the url **https://davesteele.github.io/comitup/latest/davesteele-comitup-apt-source_latest.deb**  from the install-comitup.sh - I get also the 404-Error

Can you please check the url for comitup?!

Regards
Rolkow

outdoorbits, 2024-07-13:
Thank you for this information. I'll adapt the installation process next week. If you need a quick solution, you could follow the new installation instructions of comitup.

outdoorbits, 2024-07-14:
It's solved. Just install or update lbb.

closed 2024-07-15 by outdoorbits