outdoorbits/little-backup-box #315: Update via SSH from Windows Gets Wrong Keyboard Inputs

macmcb, 2024-07-20:
When I try to update LBB via SSH from Windows 10, the resulting screen does not interpret keyboard inputs correctly. Example (see screen image): I hit Up-Arrow key and "^[[A" appears on the screen (instead of moving the selected item up). I get the same problem using SSH via both Powershell and Cmd.exe)

The pi is connected by wifi, and I am logged in as user "pi". I then run the installation command provided in the installation instructions. In this case, I am updating--not clean install--but you have said this same command should work in update mode if LBB is already installed. 
`branch='main'; curl -sSL https://raw.githubusercontent.com/outdoorbits/little-backup-box/${branch}/install-little-backup-box.sh | bash -s -- ${branch} 2> install-error.log`

Result after I press a few up- and down-arrows and number keys:
[image]

 - OS: Windows 10 Home
 - Raspberry Pi 4B

outdoorbits, 2024-07-21:
The keyboard issue should not be relevant because the asked decision should never pop up. The decision should be forced to use the old config.
I just ran the update/install command without any issue. Please provide more information. Are you using bookworm? Are there any other modifications on this system excepted lbb?

macmcb, 2024-07-21:
Running Bookworm and LBB. No other modifications. Only changes to settings via the LBB Settings screen.

outdoorbits, 2024-07-21:
Could you publish your install-error.log?

outdoorbits, 2024-07-21:
... and did you try to update via web UI (Settings - update)?

macmcb, 2024-07-21:
Thanks for the assistance. 

install-error.log is empty ($ cat install-error.log and $ nano install-error.log).
The date on the file is 13 July ($ ls -l). Not sure, but this could have been the last time I tried to run Update from the Settings page.

Just tried Settings > Update again. The inset frame shows 
> Internal Server Error
> ... 
> More information about this error may be available in the server error log.
> Apache/2.4.59 (Debian) Server at raspberrypi.local Port 80

outdoorbits, 2024-07-21:
I can not reproduce this at all. Strange. Maybe there is any damage on your installation. Did you try a completely new installation, maybe on another SD card? I can update here both ways without any problems and I'll do another new installation test later.

outdoorbits, 2024-07-21:
I can not reproduce this at all. Strange. Maybe there is any damage on your installation. Did you try a completely new installation, maybe on another SD card? I can update here both ways without any problems. A new installation ran just now without any problems.

macmcb, 2024-07-21:
It makes sense that the problem has something to do with my system or LBB settings--something that changed after the initial install.  I will do a clean install and report if/when the update problems happen again.

outdoorbits, 2024-07-21:
Please report in any case.

macmcb, 2024-07-22:
Clean install on a different SD card seems to be working fine. My keyboard inputs worked as expected. I changed a few settings in the LBB web interface, then issued the install command again via SSH,  and it ran in update mode as expected. I plan to reformat and reinstall LBB on the problem SD card to test and possibly use as a backup when I travel. I think you can close this issue. I'll let you know if the problem happens again.

As always, thank you for your help and for LBB.

outdoorbits, 2024-07-22:
Very good! Having at least a second working SD card is always a good idea. In bad luck situation linux could be not bootable anymore after a simple power cut. You might be luck if you just have to exchange the card ...

closed 2024-07-23 by outdoorbits