outdoorbits/little-backup-box #317: No thumbnails

ruebyi, 2024-07-30:
Hi, 
I‘m traveling right now, and I don‘t get any thumbnails generated of my ORF Files from my OM-1.

[image]

outdoorbits, 2024-07-30:
I found there is an issue about just this type of files in libraw 0.20 
(actual bookworm version). But there is an updated version having 
positive reports. Let me see if I can fix this.

ruebyi, 2024-07-30:
Thank you!
Same form raw files from my OM-5

ruebyi, 2024-07-30:
There is a snapshot with support for Om-5
https://www.libraw.org/news/libraw-202403-snapshot

ruebyi, 2024-07-30:
And 0.21 seems to support OM-1

outdoorbits, 2024-07-30:
I hope my solution works for you. Please

1. update to actual version
2. In Settings - Update klick the button 'Update libraw to 0.21.2-2.1'
3. You have to remove all the dummy files from tims folders (thumbnails)
4. Create thumbnails again
5. Tell me if it is working ;-)

ruebyi, 2024-07-30:
Than you

update I stuck…

[cmd 3.log](https://github.com/user-attachments/files/16433075/cmd.3.log)

outdoorbits, 2024-07-30:
Did you try to create thumbnails from raw? Maybe the last lines of the update just didn't show up in log?

ruebyi, 2024-07-30:
Not yet - I can‘t find the dummy files - ca I access them via file browser of lbb?

outdoorbits, 2024-07-30:
The files into the tims folders below the original images. Every original image has a thumbnail here. Because the original image could not be used, there are alternative grey images I called dummies.

ruebyi, 2024-08-01:
It worked for my om-5 files, but not for my Om-1…

outdoorbits, 2024-08-01:
But is there a difference to before the modification?

ruebyi, 2024-08-01:
Yes OM-5 Files get a preview now

outdoorbits, 2024-08-01:
This is strange. In the compatibility list of libraw version 0.21 you can find "OM Digital Solutions OM-1" (since 0.21.0) but not OM-5. It should be vice versa.
OM-5 should be supported sice LibRaw 202403 snapshot, which is not available as package but as source code only.

ruebyi, 2024-08-01:
I just deleted the Tim’s folder completely in the OM-1 folder and rebuild previews. 
it‘s a very long process on a pi Zero W

ruebyi, 2024-08-03:
It worked! Took my 30hours, but it created all 3000 thumbs

outdoorbits, 2024-08-03:
Great, have a nice trip!

closed 2024-08-03 by outdoorbits