outdoorbits/little-backup-box #344: Backup gets stuck at 99.8%

norbertborsky, 2024-12-28:
**Describe the bug**
When i try to back up from usb to internal storage (microSD card) the backup process always gets stuck at 99.8% (last file). I have tried to let it go for over an hour and nothing happens and tried it multiple times. I use Raspberry Pi 5, Belkin Devices card reader via usb cable. 
I have tried already to make a completely fresh install of Raspberry Pi but the problem persists.

**To Reproduce**
Steps to reproduce the behavior:
1. Back up from USB to internal storage

**Expected behavior**
Finishes the backup.

[image]

**Logs**
`2024-12-28 18:48:13
Ready
Insert sources

2024-12-28 18:48:13
pre mount usb, source (device list changed)
--uuid 6657-E919

2024-12-28 18:48:13
DeviceChosenIdentifier='--uuid 6657-E919' ('/dev/sdc1') prepared to mount at '/media/usb_source', exfat

2024-12-28 18:48:13
MOUNTED?: '/media/usb_source' = False

2024-12-28 18:48:14
mounting:
USB

2024-12-28 18:48:14
mount --uuid 6657-E919 /media/usb_source -o uid=33,gid=33,umask=0

2024-12-28 18:48:14
mounted USB '--uuid 6657-E919' at '/media/usb_source': Msg.=''

2024-12-28 18:48:14
MOUNTED?: '/media/usb_source' = True

2024-12-28 18:48:14
Source o.k.
size: 250G
used: 106G
free: 143G
FS: exfat
 42.6% >>>>>>>>____________

2024-12-28 18:48:14
mount USB device: --uuid 6657-E919

2024-12-28 18:48:14
MOUNTED?: '/media/usb_source' = True

2024-12-28 18:48:14
Backup from usb: 

2024-12-28 18:48:14
Lost target device internal? False

2024-12-28 18:48:14
MOUNTED?: '/media/usb_source' = True

2024-12-28 18:48:14
Lost source device usb? False

2024-12-28 18:48:14
Working...

[... 1322 lines cut ...]

2024-12-28 19:11:46
USB storages 
 > Int. storage 
522 of 526, 6.332Mi/s
Time: 0:00:10
 99.2% >>>>>>>>>>>>>>>>>>>_

2024-12-28 19:11:48
USB storages 
 > Int. storage 
524 of 526, 6.332Mi/s
Time: 0:00:05
 99.6% >>>>>>>>>>>>>>>>>>>_

2024-12-28 19:11:57
USB storages 
 > Int. storage 
525 of 526, 6.332Mi/s
Time: 0:00:02
 99.8% >>>>>>>>>>>>>>>>>>>_

pi@lbb:/var/www/little-backup-box/tmp $ date
Sat 28 Dec 19:37:53 CET 2024
`

outdoorbits, 2024-12-28:
There was a bug using the wrong tool for this backup. It used rclone which is for cloud backups. Now that if fixed. Would you please update and tell me? In my test it worked now without a problem. (But before the changes it didn't stop like it was in your case).

outdoorbits, 2024-12-28:
Sorry, now it is ready for update.

norbertborsky, 2024-12-29:
Thank you for the quick reply. I have tried 2 times and now it always gets stuck at 52.6%. Also second issue i wanted to report, when i open the rclone gui it's just a blank white screen (tested on multiple devices).
[image]
[image]
[image]

norbertborsky, 2024-12-29:
One more update. I let it go for a longer time and after something over 30 mins doing nothing it continued to work. Is that a normal behavior?
[image]

outdoorbits, 2024-12-29:
At first the rclone gui problem: Firefox does not allow to access the embedded rclone gui window for security concerns and as I found now, Chrome does also not allow this anymore. You get there by this link: http://111.222.333.444:81/ (use your IP).
I'm working on that but I'm still facing problems in apache2 webservers configuration. If you or anybody else has this skills, your help would be very appreciated.

The delay of more than half an hour is interesting. The process works after that as expected? Is there maybe a extremely huge file (~70 GB)? Can you copy all the files from your usb to your PC or is there an error message (maybe a corrupted file)?

What's about your free disk space on your SD? Keep in mind, if you have no space left, even the operating system might get unstable. I would recommend not to backup to internal if you have different options.

norbertborsky, 2024-12-29:
After that pause it continued to work and finished the backup. I include the full log to investigate, the most interesting part is between 9:27 and 10:04 (as you can see it has skipped two files in the log). There is no larger files then a 300MB video.

Regarding the storage, it's a 256gb sd card so there is plenty of room. I plan to install an NVMe drive to use it for the backup, it was just a first testing and found out this bug. I will do some more testing next week.

[little-backup-box.log](https://github.com/user-attachments/files/18269056/little-backup-box.log)
[image]

norbertborsky, 2024-12-29:
Also wanted to mention, I truly appreciate your quick responses and eagerness to assist. Thank you! :)

outdoorbits, 2024-12-29:
Thank you ;-)
File #278 to #280 does not mean a skip. This is just an issue of not logging every single transfer. It counted #279 silent.

outdoorbits, 2024-12-30:
Thanks to the help of the ai Claude the web servers config now is working and rclone gui as well as comitup are displayed as expected.

norbertborsky, 2024-12-30:
looks good! :) i will do some more testing next week with proper hw, this original reported bug is solved so we can close the issue i guess.

closed 2024-12-30 by outdoorbits