outdoorbits/little-backup-box #346: Missing comitup wifi configuration

norbertborsky, 2025-01-01:
Hi, it's me again :) 

Seems like we have lost in the last update the comitup wifi configuration page. It was working but after the new update,  when i go according to the documentation to http://10.41.0.1 (when there is no known wifi), it goes directly to LBB gui. 

Right now i was not able to find any way set up a wifi connection, only to connect directly to the LBB in hotspot mode. If i go to comitup from the GUI, it just shows the whole gui once again just embedded. I have tried in on different machines (WIN, MACOS, Android) and different wifis. Also tried to make a clean installation of the Raspberry Pi but the problem persists.

[image]

outdoorbits, 2025-01-01:
In case comitup can not connect to a network, it publishes the comitup frontend. If the frontend is up, the LBB GUI activates the link to comitup on the top of the page.

Could you check if the link is active? If not, go into settings and reset comitup. After a few seconds reload the GUI and check the link. You should reach comitup by
http://111.222.333.444 or 
https://111.222.333.444/comitup/ (the final slash is important.)

outdoorbits, 2025-01-01:
Oh, sorry, I didn't see the image. Is this really the actual version? I'm irritated because of the active comitup link. The LBB GUI inside the frame means, the link should be inactive because the comitup frontend is down. Would you please reset comitup by settings?

norbertborsky, 2025-01-01:
Yes, it's the most recent version (as i mentioned previously, i also made a completely clean install in order to try to fix it). 
This behavior started with the version that has fixed the rclone gui. 

I have tried to reset the comitup settings now and it's the same. There is no show up of comitup frontend to connect to any wifi that used to be there before the rclone fix update.

outdoorbits, 2025-01-01:
I have to reproduce this tomorrow. Do use have a second wifi adapter in place or are you just using the internal of your pi?

norbertborsky, 2025-01-01:
i use the internal one.

norbertborsky, 2025-01-02:
I have tested it and works like a charm :) Thank you once again for the great work!

outdoorbits, 2025-01-02:
Thank you! Testing in different environments is very important and improves the project bit by bit...

closed 2025-01-02 by outdoorbits