outdoorbits/little-backup-box #349: Tiny File Manager asking for login credentials

brainhig, 2025-01-22:
**Describe the bug**
Tiny File Manager asking for login credentials after changing option to show hidden files and saving (changed and saved twice as it didn't take the first time.

**To Reproduce**
Installed from scratch on new Rasberry Pi OS (bookworm). Installed LBB. Backed up one SD Card to external SSD. Went into Files, turned on show hidden files. Redirected to login page for Tiny File Manager. No idea what those credentials are, doesn't work with Pi login or admin/admin

**Expected behavior**
Return to options page or main file manager page, not login page.

**Desktop (please complete the following information):**
Chrome browser on Windows

brainhig, 2025-01-22:
Found default credentials via Google search. For anyone else that finds themself in this predicament, the credentials are:

admin
admin@123

outdoorbits, 2025-01-22:
This looks like a bug in tinyfilemanager. I'll try to reproduce it within the next days.

outdoorbits, 2025-01-22:
I already created an issue for that in 2024/10 and now I did it again. Last time I didn't get any response...

brainhig, 2025-01-24:
Thank you for the response. I'll give the change a try and let you know :)

outdoorbits, 2025-01-26:
If you are using the final version, you can just change the language in **lbb** settings to overwrite the wrong tinyfilemanagers config.php.

brainhig, 2025-01-27:
Yep, seems to work just fine. I installed from scratch on another Pi, and it was fine. And I edited the tinyfilemanager config on my original install and that worked as well. 

Thank you for your help!

closed 2025-01-27 by outdoorbits