outdoorbits/little-backup-box #351: Sync database after backup hangs forever

brainhig, 2025-01-30:
**Describe the bug**
After backup process copies all images from connected USB source, sync database starts, but hangs. 

**To Reproduce**
Connected target and source, manually selected target and source. Begin USB to USB backup

**Expected behavior**
Sync database process to complete and report that it is completed.

**Screenshots**

[image]

Log output:

Little Backup Box
2025-01-30 11:10:48
IMAGE=/var/www/little-backup-box/little-backup-box.bmp

2025-01-30 11:21:55
Backup
aborted

2025-01-30 11:21:55
Preset source: --uuid 2F37-6736

2025-01-30 11:21:55
Preset target: --uuid E216-576D

2025-01-30 11:21:55
USB storage 
 > USB storage 

2025-01-30 11:21:55
Source: usb 

2025-01-30 11:21:55
Target: usb 

2025-01-30 11:21:55
mount device: usb, target; WaitForDevice=True; DeviceIdentifierPresetThis=--uuid E216-576D; DeviceIdentifierPresetOther=--uuid 2F37-6736

2025-01-30 11:21:55
MOUNTED?: '/media/usb_target' = False

2025-01-30 11:21:55
umount /media/usb_target: Not mounted.

2025-01-30 11:21:56
Ready
Insert target

2025-01-30 11:21:56
pre mount usb, target (device list changed)
--uuid E216-576D
--uuid 9A73-C0F3
--uuid EC8C-2C6E

2025-01-30 11:21:56
DeviceChosenIdentifier='--uuid E216-576D' ('/dev/sda1') prepared to mount at '/media/usb_target', exfat

2025-01-30 11:21:56
MOUNTED?: '/media/usb_target' = False

2025-01-30 11:21:56
mounting:
USB

2025-01-30 11:21:56
mount --uuid E216-576D /media/usb_target -o uid=33,gid=33,umask=0
[... 2330 lines cut ...]
USB storage
22 of 336
Time: 0:02:22
 6.5% >___________________

2025-01-30 11:28:15
Sync database
USB storage
24 of 336
Time: 0:02:21
 7.1% >___________________

2025-01-30 11:28:15
Sync database
USB storage
25 of 336
Time: 0:02:23
 7.4% >___________________

2025-01-30 11:28:16
Sync database
USB storage
27 of 336
Time: 0:02:23
 8.0% >___________________

brainhig, 2025-01-30:
Follow-up:

I wiped the USB target SSD and all databases, performed backup again from Sony camera USB connected. Syncronization and thumbnail generation completed properly.

I had some images/videos from DJI Action 5 on the target SSD, I suspect those files were causing the hang. I will backup the files from the DJI and see if that completes.

outdoorbits, 2025-01-30:
At first, full logs of the complete progress are hard to read. Please try to shorten them like I did above ;-)

I can not say what's the problem right now. Coud you login to your pi and execute this command:
`sudo python3 /var/www/little-backup-box/backup.py --SourceName database --TargetName usb --power-off False`
Hopefully you get an error message. Please post it here.

outdoorbits, 2025-01-30:
> I had some images/videos from DJI Action 5 on the target SSD, I suspect those files were causing the hang. I will backup the files from the DJI and see if that completes.

This would be very interesting. If you find out the files causing the error I would like to examine the error with your help.

brainhig, 2025-01-30:
I connected a card reader with the DJI SD card inserted to try the backup that way. (I checked the integrity of the SD card beforehand). The backup process transfers all of the images and videos, of which there are 23, as well as the thumbnails DJI generates which are very small, hence how quickly they are transferred. The log shows this before unmounting the USB source:

2025-01-30 13:11:13
USB storages 
 > USB storage 
23 of 54, 18.81MB/s
Time: 0:01:46
 42.6% >>>>>>>>____________

2025-01-30 13:11:13
USB storages 
 > USB storage 
54 of 54, 18.80MB/s
Time: 0:00:00
 100.0% >>>>>>>>>>>>>>>>>>>>

After which, the sync hangs again:

2025-01-30 13:56:59
Sync database
USB storage
Counting images.
Just a moment...

2025-01-30 13:56:59
Sync database
USB storage
0 of 318
Time: ?
comparing

After rebooting, I tried a sync and clean database which then hangs again:

2025-01-30 13:29:41
Cleaning database
USB storage
300 of 300
Time: 0:00:00
 100.0% >>>>>>>>>>>>>>>>>>>>

2025-01-30 13:29:41
Sync database
USB storage
Counting images.
Just a moment...

2025-01-30 13:29:42
Sync database
USB storage
0 of 318
Time: ?
comparing

After rebooting, I deleted the folder on the target that had just been backed up, and the database sync completed immediately.

2025-01-30 13:46:00
Sync database
USB storage
300 of 300
Time: 0:00:00
 100.0% >>>>>>>>>>>>>>>>>>>>

2025-01-30 13:46:00
finished

After another backup, I deleted the extra files from the backed up SD card directory (thumbnails and index) and tried the sync again, but again it hangs. Something about the files from the Action 5 Pro? Including a screenshot of the folder:

[image]

brainhig, 2025-01-30:
After more testing, it looks like it is the JPG files that are causing the issue. I removed each type of file, one by one, followed by reboots and sync tests, leaving only the JPG files. The sync process still hangs. Removed the remaining JPG files and the sync process completes. 

Not sure how to test these JPGs to see where the failure point is. Something to do with EXIF data? During the sync process, do the images get scanned by ffmpeg? 

Let me know if there is something I can do to troubleshoot.

I appreciate your responses and help!

brainhig, 2025-01-30:
Here is the output from the command you provided:

pi@lbb:/var/www/little-backup-box/tmp $ sudo python3 /var/www/little-backup-box/backup.py --SourceName database --TargetName usb --power-off False
Traceback (most recent call last):
  File "/var/www/little-backup-box/backup.py", line 1557, in <module>
    backup_primary.run()
  File "/var/www/little-backup-box/backup.py", line 218, in run
    self.syncDatabase()
  File "/var/www/little-backup-box/backup.py", line 1060, in syncDatabase
    db.dbInsertImage(FileName)
  File "/var/www/little-backup-box/lib_view.py", line 200, in dbInsertImage
    self.__cur.execute(f"alter table EXIF_DATA add column '{EXIF_Field}' text;")
sqlite3.OperationalError: duplicate column name: Gps_Status

outdoorbits, 2025-01-31:
Fantastic! So some of your files have an exif information twice. This causes the error because the script doesn't handle this unexpected situation. I'll fix this soon, hopefully today or tomorrow.
Thank you for debugging, great job!

outdoorbits, 2025-01-31:
The update is online, please just use the web UI based updater. I'm not sure if the problem might persist because the code looked to be proof to that error. If it persists, could you please mail such a problematic image? You find my mail in nearly any code file.

brainhig, 2025-01-31:
Thank you so much for looking at this!

I see in the database that the scan process for the Sony images adds the column "GPS_Status" and the error message refers to "Gps_Status" Seems like when trying to add the column, sqlite is case insensitive when evaluating a column name?

As a workaround, I declared a lowercase version of the array "self.__EXIF_KnownColumnsList" and compare EXIF_Field.lower() to that in line 199:

` if not EXIF_Field.lower() in known_columns_lower:`

This avoids adding the column if the column name is the same but the case is different.

After this change, the database sync runs successfully.

outdoorbits, 2025-01-31:
Maybe I found a possible error. The exif entry for 'Gps_Status' is duplicated in your file. Maybe this is not filtered, because the entries differ in case like 'Gps_Status' and 'GPS_Status'. Now I check case insensitive. Could you try again? If the error persists, I think I really need an image.

brainhig, 2025-01-31:
After updating to your new code, I get a different error now :( This happens on the first image scanned, after deleting the database and re-running the backup.py command.

pi@lbb:~ $ sudo python3 /var/www/little-backup-box/backup.py --SourceName database --TargetName usb --power-off False
Traceback (most recent call last):
  File "/var/www/little-backup-box/backup.py", line 1557, in <module>
    backup_primary.run()
  File "/var/www/little-backup-box/backup.py", line 218, in run
    self.syncDatabase()
  File "/var/www/little-backup-box/backup.py", line 1060, in syncDatabase
    db.dbInsertImage(FileName)
  File "/var/www/little-backup-box/lib_view.py", line 203, in dbInsertImage
    self.__cur.execute(f"alter table EXIF_DATA add column '{EXIF_Field}' text;")
sqlite3.OperationalError: duplicate column name: ExifTool_Version_Number

outdoorbits, 2025-01-31:
self.__EXIF_KnownColumnsList needs to append EXIF_Field.lower(), not EXIF_Field. In my test I could reproduce the crash after removing the database. Now in ran smoothly.

outdoorbits, 2025-01-31:
Hopefully the now uploaded version will work.

brainhig, 2025-01-31:
Perfect! I made that last change in lib_view.py and it ran without a problem :)

Thank you so much for all your work on this whole project!

closed 2025-01-31 by outdoorbits