outdoorbits/little-backup-box #352: libraw 0.21.2-2.1 no longer available in debian repository

brainhig, 2025-01-31:
**Describe the bug**
Try to update libraw to 0.21.2-2.1 in order to read Sony A7-IV .arw images

**To Reproduce**
Go to System, Update, Update libraw to 0.21.2-2.1

**Expected behavior**
Update to newer libraw 

**Screenshots**

[image]

**Additional context**

Modified update_libraw.sh to download available .deb packages version 0.21.3-1+b1. However, libc6 version >2.38 is required, so no update is possible to a version that will support Sony A7-IV

outdoorbits, 2025-01-31:
I tried to install libc6 and libraw from unstable but without any success. If you have any solution or idea, please let me know.

brainhig, 2025-02-01:
Thank you for checking it out :) I didn't try to install libc6 on my Pi, since I've got it working well now, and I'm going on holiday soon and will be relying on LBB for a week :) However, if you do have access to those 21.2 .deb files, I suppose I could update the packages. I haven't tried to find them online anywhere, and am wary of installing them if I did!
 
When I return I'll spend some time on it. It's not of utmost importance, since the RAW files are being backed up successfully, and I don't REALLY need to see those thumbnails.

outdoorbits, 2025-02-01:
> The Message ID: 
> ***@***.***>

The files are available also here:

http://ftp.debian.org/debian/pool/main/libr/libraw/libraw-bin_0.21.3-1+b1_arm64.deb
http://ftp.debian.org/debian/pool/main/libr/libraw/libraw23t64_0.21.3-1+b1_amd64.deb

but it's the same problem about dependencies. And there are much more 
problematic dependencies of the required libc6.

I hope you have a nice vacation. Please remember to have a second 
installation on an SD card with you in case the installation no longer 
boots after a power failure, for example. 
https://github.com/outdoorbits/little-backup-box/wiki/06.-Thoughts-for-travelling

closed 2025-02-11 by outdoorbits