outdoorbits/little-backup-box #365: Optional checksum / verification of the copied files

climbingpixeljuri, 2025-03-19:
**Is your feature request related to a problem? Please describe.**
To be able to be sure that all the pictures / video files are copied, it would be amazing to have some sort of Checksum / verification of the copied files. 

**Describe the solution you'd like**
An optional setting under modifications for XXH64BE checksums.

outdoorbits, 2025-03-20:
Could you describe this more specific? Do you think about a checksum sent in the mail report or do you just want to use a checksum while transfer?

climbingpixeljuri, 2025-03-20:
> Could you describe this more specific? Do you think about a checksum sent in the mail report or do you just want to use a checksum while transfer?

It is a feature used by other backup software like ShotPut Pro or OffShoot. Usually after all transfers are finished, the software creates checksums of each source and copied files and compares them. Of course no changes to the files can be allowed for this kind of backups (name or metadata change, compression etc). This way I can do the backup of an SD card on location and be sure no errors occurred during the file transfer. 

So it would be great to have an option to turn on verification checks that start directly after the copies are done and show some sort of verification in the end (via email oder in the log monitor).

Thank you for your work.