outdoorbits/little-backup-box #368: At end of successful backup, the log file shows "11 of 0 files copied" instead of the total files to be copied

jconno03, 2025-03-28:
lib_backup.py has the following on line 489
self.display_summary.append(f":{FilesProcessed} {self.__lan.l('box_backup_of')} {**FilesToProcess**}{'+' if Report['FilesToProcess_possible_more'] else ''} {self.__lan.l('box_backup_files_copied')}")

At the end of a backup, FilestoProcess=0.  I think this should be using on FilesToProcessReport which has the correct value.

outdoorbits, 2025-03-28:
I do not understand. The message is, how many files are processed and how many have been expected to process. Why should this second value be 0?

outdoorbits, 2025-03-29:
What transfer type did you use? In your case the calculated file number was zero but 11 files where transferred. Please describe all the process and post the log.

jconno03, 2025-03-31:
I was copying USB to USB.  In my case, from a dive camera to a USB stick to verify an upgrade of the Raspberry Pi and the backup software.  I did not save those test logs.  I did run a test with the change to lib_backup.py as noted above and I got the results I wanted, which is 11 of 11 files copied.   A minor issue but I put the log on a lcd display so it is a visible issue to me.

outdoorbits, 2025-04-07:
This difference means, the estimated number of files to transfer was 0, but in real it were more. As this was a transfer from camera to memory stick, I expect it was executed by gphoto2. Do you remember the messages during the backup? Did it expect 0 files? Could you try to reproduce this and send the log file?
FilesToProcessReport means the files to process in one (of maybe many) backup folder. In line 
`477 FilesToProcess	+= FilesToProcessReport `
they are summarized into FilesToProcess.
Could you please try to reproduce this and share the log here?

closed 2025-04-28 by outdoorbits