outdoorbits/little-backup-box #372: Another Display

thelightdecides, 2025-05-02:
Hello, 
I would like to use the 0.96 LCD display from waveshare, as it has a lower installation height. The driver should actually be the existing ST7735. But I don't get a picture. What is not correctly selectable is the resolution (160x80), maybe this is the problem. The display itself works, I tested it with the demo from waveshare.
If it is not too much effort, it would be nice if you could get it to work. Unfortunately I can't help with this myself, I'm not familiar with programming.

https://www.waveshare.com/0.96inch-lcd-module.htm

outdoorbits, 2025-05-02:
Just update, it's there. ;-)

thelightdecides, 2025-05-03:
Thank you very much. The display is now working. But the picture is shifted downwards. The correction in the settings of +5/-5 is unfortunately not enough to correct this.

[image]

outdoorbits, 2025-05-03:
I increased the range to +-10 now. Is this sufficient?

thelightdecides, 2025-05-03:
It's better, but not perfect yet, I think at least 15, better 20.

outdoorbits, 2025-05-03:
I increased the range to 20. But I'm wondering about that. Please give feedback and tell me, if all the screen is used.

thelightdecides, 2025-05-04:
I took another look at the Wiki for the display. It actually says that an offset of 27 pixels is necessary.
“Since the display starting position of the LCD is inconsistent with the origin of the controller, it is necessary to perform offset processing when using the controller to initialize the full-screen display area: the display starts from the second pixel in the horizontal direction, and the 27th pixel in the vertical direction. ”
Sorry, I should have found that before.
Another problem is, the background is white, the font is relatively hard to read, especially with some of the colors, gray works. Is there a way to make the background black? I also had to disconnect the cable for the backlight from PIN 18, contrary to the instructions for the display, or put it on PIN 24 (this is in other instructions for waveshare displays). If the cable makes contact with PIN 18, the whole display remains black, no characters can be seen. I don't know if this might be causing the problem with the light background.

I'm still waiting for a different OLED display, which is a bit smaller. Maybe that will cause fewer problems. So if the changes are a lot of work, we could wait. I wanted to build a new case and need a display that is as low as possible, the standard displays always take up unnecessary space for the jumper cables.

outdoorbits, 2025-05-04:
The offset is increased to +-30 now. This display looks very interesting to me, so please don't give up. Managing the background color should not be a very big deal, but could take a time. Also managing the backlight might not be too complicated. I will read the docs also...
At the end we should not forget to add a wiki for this special settings.

outdoorbits, 2025-05-04:
I couldn't keep my hands off it. Now you can adjust the background color, the backlight PIN, and its state in the setup. Does it work now?
Where did you find the offset specifications?

thelightdecides, 2025-05-04:
I now have a full screen. I can change the background color. But it is always the opposite color (complementary). If I choose white, then the background is black, the same is with the font colors, if I choose blue, the font is yellow (but that was already the case before). No big deal. 
When I activate the backlight, the screen turns black, nothing can be seen. When I remove the plug from the pin, everything is there again. So the exact opposite seems to happen here too.
But all in all, it works now! Thank you very much!!!

I found the info here: https://www.waveshare.com/wiki/0.96inch_LCD_Module      "LCD and controller"

outdoorbits, 2025-05-04:
Crazy. I tested the colors here, but on a SSD1331 device, which has no 
backlight. The colors where correct. I just ordered this device. It is 
very interesting to me because of its better resolution. We will find 
out...

outdoorbits, 2025-05-05:
It looks like there are variants of this driver (ST7735 and ST7735S?). The color can be RGB or BGR coded, that's likely why you get inverted colors.
Last changes:

- moved backlight-PIN-configuration to hardware setup
- added option "Use inverted colors (BGR instead of RGB)" to hardware setup

As I could not test this yet, I'm excited about your testing.

I also added a wiki article: [https://github.com/outdoorbits/little-backup-box/wiki/02a-1.-ST7735-Devices](https://github.com/outdoorbits/little-backup-box/wiki/02a-1.-ST7735-Devices) Would you please check this also?

thelightdecides, 2025-05-05:
That hasn't changed much. The colors are still different. 
RGB setting: Font color yellow, font is red
BGR setting: font color yellow, font is dark blue
But that's no problem for me, I'll choose suitable colors once and then not change them anymore.
I'll read the Wiki tomorrow.

thelightdecides, 2025-05-05:
Ah, wiki was short ;). Yes its ok.

But:
"GPIO pin (BCM notation) is connected to the backlight: The manual suggests GPIO 18 (PIN 12)"
If you connect it, you have to turn backlight OFF. If you turn ist on, the whole screen will be black.

outdoorbits, 2025-05-06:
I accepted the manual because that issue happened when the backlight pin was not yet managed at all. But I will  change this.
Does the backlight setting on/off work?

outdoorbits, 2025-05-06:
Does this display depend on the backlight to make anything visible?
I am still not fine about the wrong colors. Maybe I fixed that now.

thelightdecides, 2025-05-06:
Yes, it needs a backlight. Turning it on/off works, but it works inverse, when I turn it on in the lbb the display went black, when I turn it off the display works. it also works if i disconnect the cable for the backlight.

outdoorbits, 2025-05-08:
What's about the colors in the actual version, are thy fixed?

thelightdecides, 2025-05-09:
No, unfortunately not.

thelightdecides, 2025-05-21:
I am not at home at the moment. Did you try to disconnect the cable from pin 18?

outdoorbits, 2025-05-21:
This had no effect.

thelightdecides, 2025-05-21:
There are 2 st7735 driver, did you use waveshare st7735 lcd ?

thelightdecides, 2025-05-21:
I am using another display now, but i connected the lcd again and it works well.

outdoorbits, 2025-05-21:
Oh no, I expected the hat version to use other pins, it's working now. Thank you.

thelightdecides, 2025-05-21:
🙂👍

outdoorbits, 2025-05-22:
I is solved. After update there is a new option in hardware settings: Inverted colors. I also updated the wiki. For the offsets I found slightly different values (1, 26) instead of (2, 27). In the second one I had issues in the first row and the first column.

thelightdecides, 2025-05-23:
Yes, now it works perfectly. Great!
My new setup is ready. It's running without any problems so far. But i used a different display there, again an oled, which had fewer cables and they were shorter and thinner. But maybe I'll do another version with this LCD display.
My new setup: https://www.thelightdecides.com/travel-photography-blog/little-backup-box-simple-reliable-backup

outdoorbits, 2025-05-25:
Your setup is brilliant, and so is the description. Can I use images from it on GitHub? I'd like to make a clearer reference.

thelightdecides, 2025-05-25:
Thank you very much! Yes, of course you can use the pictures.

outdoorbits, 2025-05-26:
Thank you, I just uploaded the link and the images.

closed 2025-05-26 by outdoorbits

thelightdecides, 2025-05-29:
Thank you very much!