outdoorbits/little-backup-box #373: LBB switched usb_source and usb_target after they were already mounted

macmcb, 2025-05-12:
I'm using Pi 4B with a USB3 spinning HDD and CFe card reader connected via a powered USB hub.

I understand that LBB by default uses the order of connection to determine which device is source/target. I have never fully trusted that process, so I do not start a backup by default, and I always use Mount Target and Mount Source buttons on the LBB>Tools page. Once I know that the correct devices are mounted, I trigger USB > USB backup on the LBB > Backup page. 

Today's issue: 
- Both USB devices were plugged in when the Pi rebooted. 
- When I clicked Mount Target, I saw that LBB had mounted the CFe card as usb_target. Not what I needed. (This is why I always mount the devices manually.)
- I clicked Unmount Target, then removed the CFe card from the reader. HDD was still connected but not mounted.
- I clicked Mount Target again. LBB mounted the HDD as usb_target.
- I inserted the CFe card into the reader and clicked Mount Source. LBB mounted the card as expected, usb_source. 
- I triggered USB > USB on the LBB>Backup page.
- The backup finished almost instantly but processed only 9 files, copying from the almost empty HDD to the not-quite-full CFe card. (If the HDD had not been empty, it would've been much easier to not notice the problem.)

Reviewing the log, it looks like LBB failed an initial backup attempt, then unmounted both already-mounted devices and re-mounted them in the opposite roles, then successfully ran the (reversed) backup. 

I'm not sure if this is somehow expected behavior for LBB. At the very least, I would say that more user feedback is needed to clarify what is happening. Ideally, a simpler flow with less bouncing between screens in LBB. 
 
[little-backup-box(2).log](https://github.com/user-attachments/files/20164389/little-backup-box.2.log)

outdoorbits, 2025-05-12:
The trick of defining the source and target by manually mounting doesn't work. As soon as the backup starts, existing mounts are removed, and the USB drives are used in the order in which they were detected by the system. The first storage device becomes the target, all others are sources.
If multiple drives are plugged in during boot, it's probably more or less random which device is detected first and thus becomes the target.
As far as I know, the method of using the order of the storage devices as a criterion for their function works perfectly. The system assigns sda, sdb, sdc, etc. to the devices and doesn't reassign these identifiers after these devices have been removed.
However, if you really want to manually control which device should play which role, use the web interface: Backup - modifications - "Set source partition" and "Set target partition".

macmcb, 2025-05-13:
Thanks as always for the quick reply and for clarifying the logic that LBB follows. I feel like I just learned something important after more than a year of using LBB. May I suggest that this information be noted on the Tools page and clarified in the docs? 

To confirm that I understand, 
- Whichever device is sda at boot time will be the target-of-backups _any_ time that device is connected (regardless of which devices may be currently mounted as "usb_source" and "usb_target"). 
- At backup time, if the original sda is not connected, then sdb becomes the target-of-backups, and so on.
- The choices made on Backup: Modifications override the above automatic behavior for the next backup only (but choices on Tools do not).
- The "Mount Target/Source" buttons on Tools follow the automatic behavior (regardless of choices made most recently on Backup: Modifications).

To clarify this in the UI, please consider:

- On the Backup page, warn the user if they trigger a backup when there are already devices mounted as "usb_source" or "usb_target". If possible, detect whether the mounted device matches the one that the backup would use. For example "A different partition is currently mounted as "usb_target" (partition sdb1, UUID XYZ123, 1.8 TB, exfat, "label"). Which device do you want to use as the Target in this backup? [ Use Currently Mounted usb_target as Target (sdb1) ]  [ Use the Partition Selected by Little Backup Box (sda1, UUID ABC456, 110 GB, exfat, "label") ] "
- On the Backup page, notify the user if the original (or previous?) target partition is changed for a backup. In other words, "The previous Target partition was not found (sda1, 1.8 TB, exfat,  "label"). The Target partition for this backup is now: sdb1, 321 GB, exfat, "label". [ Go Ahead ] [ Cancel ].   
- On the Tools page, clarify for users that "source" and "target" are chosen automatically. "Mount Source/Target" does affect which partitions are used as source/target for subsequent backups, and does not obey choices on Backup: Modifications.
- On the Tools page, consider labeling the Mount... buttons with something other than source/target. Or include the partition, UUID, size, label in addition.

outdoorbits, 2025-05-14:
The documentation is, of course, still very incomplete. Unfortunately, this is no different from many open source projects.
However, everyone is welcome to contribute to this project, and that includes the documentation, of course.

I want to keep the web UI as clear as possible and therefore not "burden" it with more than the bare minimum of additional information. I would have thought it self-explanatory that both the backup page (modifications) and the tools page offer the option to select the partitions to be used. Furthermore, the log monitor (or the display) provides precise instructions on what should be connected and when.

But I will definitely add to the wiki and explain the practice of assigning roles to USB drives. Although this is explained at [https://github.com/outdoorbits/little-backup-box/wiki/05.-Operation](https://github.com/outdoorbits/little-backup-box/wiki/05.-Operation), it might be a bit overlooked.

Your idea of ​​opening a dialog in case of deviations from the previous usage of the USB storage devices (in my opinion) doesn't fit with the lbb's usage concept. Backups should be performed according to strict rules and as robustly as possible, so personal monitoring of the process isn't necessary. A query could stop the backup and thus negatively impact data security because an expected backup might not be performed.

closed 2025-05-21 by outdoorbits