outdoorbits/little-backup-box #379: Problem with install (password after install)

VVilku, 2025-07-15:
I'm having some sort of problem with the installation. 

I install RPi OS 64 bit (or 32 bit) as described, give password for pi user and enable ssh. 
After a reboot I am able to log in. I install the drivers for the LCD-show (I have a TFT35). Then I install the LBB. After installation and reboot, I am presented with a login window. 
The password I saved for user pi does not work? 
Earlier (about 7-9 months ago) it was able to install LBB without any problem....
What am I doing wrong?

outdoorbits, 2025-07-15:
Where do you get the login screen? On your local TFT35? This local call should work without a login. I just tried this here (modified - I do not use a local TFT) and it worked. In a new installation you shouldn't be asked for a password anyway. The password for user pi is set for ssh only. So this was a clean install without any usage of old config files?

Could you login (ssh) and execute
`cat /etc/apache2/includes/password.conf`
This should be empty. If you get 

`Authtype Basic
Authname "Password Required"
AuthUserFile /etc/apache2/includes/htpasswd
<RequireAny>
    Require valid-user
    Require local
</RequireAny>`

please empty this file by

`echo '' | sudo tee /etc/apache2/includes/password.conf`

and try again (reboot).

VVilku, 2025-07-16:
I made several attempts. The same thing every time. That is until I realised that I had reflexively always updated the system (apt update/upgrade) after installing RPI OS. When on the last attempt I did not do this - the problem was solved. I don't have a login window and LBB starts up without a problem.

VVilku, 2025-07-16:
I even tried installing without HDMI plugged in, just connectivity via ssh. LCD35-show + LBB. The effect was the same. On the TFT I have some leftover from booting, and on HDMI you can see the LBB.

VVilku, 2025-07-16:
I keep trying from the beginning. Again and again, different version etc. :) At the moment I have a version that on the TFT I have rc-local.service screen locked, and when I connect the monitor via HDMI I have LBB (working). I installed via SSH, without HDMI. 

I can package this image and post it somewhere. (?)
I think the key here now is this 35" screen and driver issues. 
If you don't have a TFT like that then you won't reproduce this.

outdoorbits, 2025-07-16:
I think you are right, most probably it's an issue about the screen driver. But I do not understand that login screen. Could you take a photo?

closed 2025-07-22 by outdoorbits